EPFL-LAP/dynamatic #370: [Buffer] Explicitly Specify Required Buffers in Op Declarations

shundroid, 2025-03-29:
In PR #356, internal buffers in ops like `Merge` and `CMerge` were removed, and their necessary buffers are now inserted by the buffer placement pass.

However, @Jiahui17 said in #342 ([this comment](https://github.com/EPFL-LAP/dynamatic/issues/342#issuecomment-2725021216)) that these internal buffers originally existed for valid reasons:

> The TEHB are for two different purposes:
> - In Merge/CMerge: avoid token reordering
> - In the load port / sharing wrapper: avoid resource sharing deadlock

Specifically, for `Merge`/`CMerge`, removing the internal TEHB can violate the handshake protocol (changing the value after setting `valid` high), as further illustrated [here](https://github.com/EPFL-LAP/dynamatic/issues/342#issuecomment-2745567695).
While I'm fine with removing the internal buffers, I believe that the requirement for such buffers is part of the op's definition and should be declared explicitly in the op definition itself, rather than hidden inside a transformation pass.

## Proposal

Instead of hardcoding required buffers inside the buffer placement pass, like in:

https://github.com/EPFL-LAP/dynamatic/blob/7d25de20cfdd8f6c15322531da5afd31fd3ff130/lib/Transforms/BufferPlacement/HandshakeSetBufferingProperties.cpp#L111-L125

I propose introducing an `OpInterface` to specify buffer requirements directly in the op declaration.
For example:

```
def MergeOp : Handshake_Op<"merge", [
  // ...
  // 1. Implements RequiredBuffersInterface with getRequiredBufferSlots
  DeclareOpInterfaceMethods<RequiredBuffersInterface, ["getRequiredBufferSlots"]> 
]> {
  // ...

  let arguments = (ins Variadic<HandshakeType>:$dataOperands);
  let results = (outs HandshakeType:$result);
  
  let extraClassDefinition = [{
    // ...

    // 2. Specifies required buffer slots per port
    unsigned $cppClass::getRequiredBufferSlots(StringRef portName) {
      if (portName == "result") {
        return 1U;
      }
      return 0U;
    }
  }];

  // ...
}
```

We can provide a **default implementation** here that always returns `0U` for the slot count, reducing code redundancy.

The buffer placement pass can then query this method via `getRequiredBufferSlots`, making the buffering requirements explicit and tied to the op’s definition.

What do you think? (CC: @murphe67 @Jiahui17 @QinYuan2000)

P.S. I said it "violates" the handshake protocol because changing the value after setting `valid` high is generally not allowed (e.g., in AXI4). But I'm not sure if this also applies to Dynamatic's handshake protocol or not. Is there a specification?

QinYuan2000, 2025-03-29:
Thank you for your proposal.

I have a question: could the return value of `getRequiredBufferSlots` be utilized within the MILP constraints?

As noted in the updated description of [PR #356](https://github.com/EPFL-LAP/dynamatic/pull/356#issue-2938137670):

>Furthermore, if the MILP solution already includes a buffer on their output channel—regardless of whether it is an OEHB, TEHB, or any other type—there is no need to insert an additional one.

This suggests that `getRequiredBufferSlots` should inform the MILP, allowing it to account for these constraints during optimization. The final buffer placement should still be determined entirely by the MILP.

By the way, `cmerge` is no longer considered part of the internal buffer issue, as noted in the updated description of [PR #356](https://github.com/EPFL-LAP/dynamatic/pull/356#issue-2938137670).

shundroid, 2025-03-30:
If you decide not to eliminate internal buffers inside Merge or CMerge, then I don't think we need to mention them in MLIR Op definition.

QinYuan2000, 2025-03-30:
> If you decide not to eliminate internal buffers inside Merge or CMerge, then I don't think we need to mention them in MLIR Op definition.

Yes, at least for now, I would treat the internal TEHB in `cmerge` as an inherent part of its functionality rather than a generic channel buffer — it serves no purpose other than ensuring `cmerge` behaves correctly. 

By contrast, the buffering required for `merge` and `mux` is more similar to the LSQ load port case mentioned in #136: an extra buffer is needed, but that buffer is not conceptually part of the unit itself.

This is my current view. Happy to hear other opinions.

shundroid, 2025-04-01:
@QinYuan2000 Why do you think `merge` is more similar to LSQ? Could you reply to [my comment](https://github.com/EPFL-LAP/dynamatic/pull/356#issuecomment-2764311502) in #342?

> Thank you for updating the pull request description!
> > As for merge and mux, unlike cmerge, their internal TEHBs can be replaced by buffers on the channel. Therefore, this PR will remove their internal TEHBs.
> 
> I'm not understanding this- I expect "As for mux, unlike merge or cmerge, " instead. TEHB inside merge is also for preventing token reordering (which happened in our case). Could you tell me why only cmerge is special?

QinYuan2000, 2025-04-01:
> [@QinYuan2000](https://github.com/QinYuan2000) Why do you think `merge` is more similar to LSQ? Could you reply to [my comment](https://github.com/EPFL-LAP/dynamatic/pull/356#issuecomment-2764311502) in [#356](https://github.com/EPFL-LAP/dynamatic/pull/356)?

Sorry for overlooking that question — I’ve just replied.

Regarding `merge`, what I meant is that the role of the buffer in the `merge` issue is similar to the LSQ case: both just need a buffer on the channel, which also does the regular job of balancing tokens. This is different from cases where a buffer is added for a special purpose (e.g. preventing token reordering) beyond balancing tokens — in those cases, an extra buffer is needed on top of what the MILP provides. (The MILP only assumes that buffers are for balancing tokens.)

If we remove `cmerge`’s internal buffer and instead add buffers on its outputs to take over its role, we end up in that case, where an extra buffer is needed beyond the MILP result (though it’s a bit tricky to explain). We want to avoid such situations — ideally, the MILP should handle all buffering needs. That’s why we decided to keep the internal buffer in `cmerge`. (Note that the internal buffer in `cmerge` does not serve to balance tokens, so it’s not considered a buffer in the MILP — it’s just part of the `cmerge` structure.)

shundroid, 2025-04-03:
I don't think `merge` is similar to the LSQ, because taking buffer out of `merge` violates the protocol. (though both can be _mitigated_ with external buffering)

I realized for `cmerge` even external buffering is technically not possible to avoid reordering issue, so this proposal would be only applied to `merge` (and `load`?).

Now I'm for keeping the internal buffer in `merge`, so I'll close this issue.

closed 2025-04-03 by shundroid