EPFL-LAP/dynamatic #373: Handshaking Protocol Specification

shundroid, 2025-04-03:
I couldn't find a formal specification of Dynamatic's handshaking protocol in the repository. Is it explicitly defined somewhere?

From what I've heard, it's based on the SELF protocol ([reference](https://www.cs.upc.edu/~jordicf/gavina/BIB/reports/self_tr.pdf)), but we use a ready bit instead of a stop bit.

This lack of specification leads to a practical issue:

- **Is the data value guaranteed to remain unchanged once the valid bit is set high?**

Eager fork (and speculator) seem to assume this, but the change in #356 removes an internal buffer from the merge operation, potentially violating this assumption.

I'm okay with not guaranteeing this behavior and instead requiring a buffer between merge and eager fork (or speculator), but it should be explicitly stated in the specification.

What do you think? @QinYuan2000 @Jiahui17 @murphe67

murphe67, 2025-04-03:
If we take the buffers out of merges, they will violate the protocol.

So we can either have fully explicit buffering or operations which fully meet the protocol, but not both.

shundroid, 2025-04-03:
I also thought this behavior was absolutely guaranteed 😂, so I actually want to keep that guarantee in the protocol.

If we do, then as Emmet said, removing buffers from merges would violate the protocol.
I don't like explicit (and external) buffering, even though I proposed it in #370, because:

- It still makes the channel between the merge and buffer invalid.
- For `cmerge`, we keep the internal buffer (since it technically cannot be replaced with an external one), so why not for merges?

If the channel between the merge and buffer is invalid, we shouldn't place anything between them, which is basically the same as having the buffer inside the merge.

QinYuan2000, 2025-04-03:
> If we take the buffers out of merges, they will violate the protocol.

I think the ultimate impact is not that the protocol is violated directly, but rather that the circuit’s throughput is reduced in order to preserve the protocol. If the throughput is very low (i.e., high initiation interval), then the next token becomes valid only after the previous one has been consumed downstream. If the throughput is high and the number of buffers is insufficient, even if the merge has an internal TEHB, the first token may occupy it, the second token can’t be consumed, and the third token arrives — leading to a protocol violation. So regardless of whether we use an internal TEHB, our goal is still to achieve optimal throughput.

> For cmerge, we keep the internal buffer (since it technically cannot be replaced with an external one), so why not for merges?

Actually, the internal TEHB in cmerge can be replaced by external buffers. Adding one extra TEHB at each output can preserve the original throughput (@Jiahui17 and I had a brief discussion on this). However, this comes at the cost of increased area utilization. (Note: by “extra TEHB,” I mean that even if the cmerge's output channel already has a buffer, we still need to add another TEHB — which differs from the merge case.) This reflects our secondary objective after achieving optimal throughput — minimizing area utilization.

My support for removing the internal TEHB in merge is not only aligned with the idea in #342 of “letting MILP handle all buffers,” but also because sometimes the output channel of a merge already has a buffer after MILP solving — making the internal TEHB completely redundant and unnecessarily increasing area usage.

shundroid, 2025-04-03:
I'm sorry but I'm not much familiar with buffering, so I want to ask you some questions. I still think preserving the protocol is more important and it doesn't affect the performance of the circuit much.

> but rather that the circuit’s throughput is reduced in order to preserve the protocol

Could you tell me how preserving the protocol affects the circuit's throughput? (I can understand there might be a bit more resource usage, but I have no idea how it affects the throughput)

> If the throughput is high and the number of buffers is insufficient, even if the merge has an internal TEHB, the first token may occupy it, the second token can’t be consumed, and the third token arrives — leading to a protocol violation

Could you tell me why this leads to a protocol violation? Does the merge not stall and miss one token? (which seems a more serious problem)

And I don't agree with any ideas to allow protocol violation. If it is needed, then the protocol should be updated and explicitly specified.

> Actually, the internal TEHB in cmerge can be replaced by external buffers. Adding one extra TEHB at each output can preserve the original throughput

Throughput is preserved, but I don't think the reordering problem is solved, is it?

> but also because sometimes the output channel of a merge already has a buffer after MILP solving — making the internal TEHB completely redundant and unnecessarily increasing area usage.

I learnt in some place you are now decrementing the number of buffers by 1 from the MLIP solution, is it correct? Is it still possible?

Thanks!

paolo-ienne, 2025-04-03:
I think this is yet another perfect topic to prepare and discuss in one of the Dynamatic Developers and Users Meetings; I fear we will not make too much progress here because I feel we are not seeing the problem the same way. If it helps, let me make a few statements.

(1) To me a MERGE is intrinsically a combinational component. In English: If there is at least one valid token at one of the inputs and the output is ready, the token is consumed and propagated. If there are two valid tokens, deterministically one of the two is consumed and propagated and the other held back. How could there be any sequential components in this?

(2) To be more specific, here is the truth table (I believe) of any MERGE. As usual, there are a gazillion of *don't cares* and an implementation would probably set them to some convenient value.

| Vin1|Vin2|Rout| |Vout|Rin1|Rin2|Sel|
|:--:|:--:|:--:|-|:--:|:--:|:--:|:--:|
|0|0|0||X|X|X|X|
|0|1|0||X|X|0|X|
|1|0|0||X|0|X|X|
|1|1|0||X|0|0|X|
|0|0|1||0|X|X|X|
|0|1|1||1|X|1|1|
|1|0|1||1|1|X|0|
|1|1|1||1|1|0|0|

The last line makes the arbitrary choice to consume the token from input 1 instead of 2. Sel controls a mux and Sel = 0 selects Data1.

(3) I cannot see how any implementation of this could per se violate the spec that we discussed above: Vout is 1 and there is a transfer only when one of the Vin1/2 is 1; if the predecessors respect the spec (keep the data while their Vout is 1), this MERGE very naturally respects the spec too.

(4) In line with probably the original intent of removing a spurious buffer (sorry, I know nothing of that problem), if there is a need of buffering, it seems a *usual* buffering problem to enable better throughput and has nothing to do with the MERGE.

What am I missing?

QinYuan2000, 2025-04-03:
Thank you all for the detailed clarifications — I now realize that my earlier focus was somewhat off-topic, and the key issue here is whether the protocol is consistently respected.

My current understanding is that a purely combinational `merge` module may change its output `data` while `valid` is high when the downstream is still not `ready`, regardless of whether the downstream component is a `fork` or something else. To eliminate this risk of protocol violation, a TEHB is needed, and the combination of `merge + TEHB` should be treated as a single unit that ensures protocol correctness from an external perspective.

Personally, I support keeping the TEHB to ensure that the protocol remains intact. Since you have more experience in defining and maintaining these specifications, I’m happy to follow whichever direction is ultimately decided and will reflect that in future implementations.

murphe67, 2025-04-04:
If the merge prioritizes input 2 over 1, it will alter the select (and therefore the output) when a token arrives at input 2 after input 1, but before input 1 is consumed, like so:

|Cycle| Vin1|Vin2|Rout| |Vout|Rin1|Rin2|Sel|
|:--:|:--:|:--:|:--:|-|:--:|:--:|:--:|:--:|
|0|1|0|0||1|0|0|0|
|1|1|1|0||1|0|0|1|

This violates the protocol.

paolo-ienne, 2025-04-04:
I do not understand: Rout = 0 so the consumer/receiver is declaring that it is not receiving the Data (*Retry* in the parlance of [this paper](https://www.cs.upc.edu/~jordicf/gavina/BIB/reports/self_tr.pdf)), so Data are a *don't care* in this case. In a sense, I see the protocol to be this:

|Data|V|R|
|:-:|:-:|:-:|
|X|0|0|
|X|0|1|
|X|1|0|
|D|1|1|

I think you are implying that the protocol is this instead:

|Data|V|R|
|:-:|:-:|:-:|
|X|0|0|
|X|0|1|
|**→D←**|1|0|
|D|1|1|

But why does Data matter if the consumer/receiver is declaring that it is not latching it? Why would the protocol specify this?

It is not a rhetorical question: I do see that this in some sense violates my answer to "Is the data value guaranteed to remain unchanged once the valid bit is set high?" but then it does not violate the only case when data matter. I guess the above table answers more precisely the original question.

Can someone show the simplest elastic circuit where a combinational bufferless MERGE with the truth table I suggested in a previous comment would lead to a malfunctioning circuit? I am unfortunately convinced that there are deep aspects these elastic components that totally escape me, so I am genuinely interested in supersimple problematic examples.

murphe67, 2025-04-04:
one option (that may be bad) that we haven't considered is to alter the merge so that it can only valid when it is ready, and this way it is both combinational and also respects the specification.

This maybe introduces headaches about deadlocking and delay propagation, and it is always risky to change a fundamental unit in case somebody relies on some aspect of it.

But it means the token propagation is always correct, and it is up the buffering algorithm to identify the tehb is needed if a merge is placed before an eager fork due to breaking a loop in the handshaking logic.

Something to think about anyway...

shundroid, 2025-04-04:
> one option (that may be bad) that we haven't considered is to alter the merge so that it can only be valid when it is ready, and this way it is both combinational and also respects the specification.

I also thought about it a bit, but it seems like it would make every eager fork behave like a lazy fork, which doesn't seem practical.

> It is not a rhetorical question: I do see that this in some sense violates my answer to "Is the data value guaranteed to remain unchanged once the valid bit is set high?" but then it does not violate the only case when data matter.

Now, I think the real question is: when does the data matter?
If data only matters when the transaction occurs, then it's the eager fork's responsibility to include a buffer.
If data matters as soon as valid is set high, then it's the merge's responsibility.

murphe67, 2025-04-04:
> I also thought about it a bit, but it seems like it would make every eager fork behave like a lazy fork, which doesn't seem practical.

I think such a merge before an eager fork deadlocks? which is even worse.

> Now, I think the real question is: when does the data should matter?
If data only matters when the transaction occurs, then it's the eager fork's responsibility to include a buffer.
If data matters as soon as valid is set high, then it's the merge's responsibility

Either is fine with me, both require buffers inside of units.

On a practical level forks are more common than merges, so I would put it in the merge.

Jiahui17, 2025-04-04:
Some units require data/valid/ready persistency in their input channel, and some units don't care. I wonder if we could specify a "needs persistent input" attribute? And units like buffer will provide a guarantee that their output is persistent (e.g., "guarantee persistent output" attribute).

The buffer placement algorithm tries to enforce certain constraints to guarantee that all persistent constraints are met. I think this also gives more flexibility to buffering as well (now you are not forced to place a buffer right after the merge, but you can also move it down in a way that all the EForks' input becomes persistent)

murphe67, 2025-04-04:
Seems like a nice way of doing it- philosophically, each unit informs the buffering algorithm of its specification out of a handful of variants?

Jiahui17, 2025-04-07:
> Some units require data/valid/ready persistency in their input channel, and some units don't care. I wonder if we could specify a "needs persistent input" attribute? And units like buffer will provide a guarantee that their output is persistent (e.g., "guarantee persistent output" attribute).

So, do you think we should go with this plan? We could split it into two steps:

1. Remove the internal buffers from Mux and Merge, and keep them in the CMerge. Add buffering constraints in HandshakeSetBufferingProperties.
2. Add a new attribute to indicate that certain units' input channels must be persistent and that certain units guarantee persistent output channels.

murphe67, 2025-04-07:
Agreed on the merge: the main benefit I see is improved resource usage by more intelligently placing the required buffer.

I think the buffer inside the cmerge still messes things up for Louis, as he would like a clean distinction between logic/buffer/arithmetic for whatever architecture he is working on- not sure this is avoidable.

shundroid, 2025-04-07:
I understand your points.
So, what do you think about putting together a quick doc outlining our handshake protocol? (e.g., channels are not persistent in some cases due to this change for some reasons, which might be actually inconvenient in some cases, and we're planning to add a feature to specify the persistency.)
Clarifying this was actually the original goal of this issue, and I now think this behavior is not trivial and especially needs explicit explanation.

As for indicating channel persistency, I think we could leverage the type system. Attributes may be harder to enforce verification.
This change is easier after #336. We can add a type parameter for specifying the persistency, just like the accompanying extra signals.