EPFL-LAP/dynamatic #378: [Integration] Managing Custom Compilation Flows Independent of Testing Framework

shundroid, 2025-04-07:
### Current Situation

Integration tests in Dynamatic can currently be run in these ways:

- Manually, via the interactive environment (`./bin/dynamatic`):

  ```
  $ ./bin/dynamatic
  > set-src integration-test/test/test.c
  > compile
  > write-hdl
  > simulate
  ```

- Automatically, using the Python script `tools/integration/run_integration.py`, or via the upcoming GoogleTest suite (to be introduced in #377):
  Both approaches internally invoke `./bin/dynamatic`:

  https://github.com/EPFL-LAP/dynamatic/blob/11079cb6bd65a8426c8c01d1eb3818842a2b95ba/tools/integration/run_integration.py#L63-L71

  https://github.com/EPFL-LAP/dynamatic/blob/7445864f271f6a5e51f7f99fd233376e6d4fe961/integration-test/util.cpp#L49-L54

### Problem

Some integration tests target specialized features (e.g., speculation) that require significantly different compilation flows (e.g., starting from handwritten IR or using different passes and backends).

These tests **cannot** use `./bin/dynamatic`, which assumes the standard flow.

In the speculation case, a separate Python script was introduced: `[run_spec_integration.py](https://github.com/EPFL-LAP/dynamatic/blob/spec-backend/tools/integration/run_spec_integration.py)` directly invokes `./bin/dynamatic-opt` with custom configuration, bypassing `./bin/dynamatic`.

The problem with this approach is that it duplicates the original integration script and makes it difficult to integrate into CI.

Once the GoogleTest suite is merged (in #377), I might add another parameterized test (`TEST_P`) for the speculation integration tests, alongside the regular parameterized tests, as suggested by @schilkp [in this comment](https://github.com/EPFL-LAP/dynamatic/discussions/314#discussioncomment-12458425) on #314.

However, there are still challenges with simply moving the logic from `run_spec_integration.py` into a custom parameterized test of GoogleTest:

- `./bin/dynamatic` remains unusable for speculation integration tests, which is unintuitive.
  - Related issue: speculation integration tests can only be executed through GoogleTest. While this is fine for CI and testing, it's not flexible or convenient for users who just want to experiment with speculation (note that the current Python script is a push-button interface).
    Users may be more interested in details like the generated circuit or initiation interval, rather than just test pass/fail status.

Therefore, I want to keep speculation compilation flow independent of the testing framework and allow test suites to call into it, just like the current parameterized tests in #377.

Note that this problem might also be applicable to @DanaKossaybati's ongoing work on out-of-order execution.

### Proposed Solution

There are several implementation strategies to support speculation integration tests once GoogleTest is integrated.

In all of these approaches, the compilation flow is separated from the test suite logic, allowing test cases to simply invoke it.

#### Option 1: Let the Test Case Call `run_spec_integration.py`

One option is to keep `run_spec_integration.py` with just the minimal logic necessary to run the speculation compilation flow, and have the test case invoke it.
This makes the speculation flow both accessible and easy to integrate with the test suite.

However, this approach still leaves `./bin/dynamatic` unusable for speculation, which can be unintuitive.

#### Option 2: Add a New Command to `./bin/dynamatic` (e.g., `compile-spec`)

Currently, `./bin/dynamatic` delegates to shell scripts using command-based dispatch (e.g., the `compile` command runs `tools/dynamatic/scripts/compile.sh` on the program set by `set-src`).

We could extend this model by adding new `compile-*` commands for each custom flow. For instance, `compile-spec` would invoke `tools/dynamatic/scripts/compile-spec.sh`.

Using parameters like `--mode spec` isn't ideal here, as most options are simply forwarded to internal scripts (e.g., `--buffer-algorithm`, `--sharing`), and using them to switch between different scripts would be confusing.

#### Option 3: Let Both Users and Test Cases Call Scripts Directly

In my view, this issue is connected to a broader design concern with `./bin/dynamatic`, described as "kind-of-a-script-but-not-a-script" by @murphe67 [here](https://github.com/EPFL-LAP/dynamatic/pull/250#issuecomment-2636786460) and also mentioned by @ebosnjak in #314:

> I'll put it this way: I currently have a Python script (`run_integration.py`) that runs an executable (`dynamatic`) that runs a shell script (`compile.sh`) that runs another executable (`dynamatic_opt`). Meanwhile, [[@shundroid](https://github.com/shundroid)](https://github.com/shundroid) has a Python script that bypasses both `dynamatic` and `compile.sh` and directly runs the required compilation flow. Honestly, this seems a bit ridiculous to me—but more importantly, it seems like another ticking time bomb in this project. I think we should start a separate discussion about this.

And I also think decomposing the monolithic `./bin/dynamatic` would make the Dynamatic ecosystem significantly simpler and easier to understand. My notion is aligned with @schilkp's idea [here](https://github.com/EPFL-LAP/dynamatic/discussions/314#discussioncomment-12460452): "Dynamatic is really more a Lego box full of parts you can use to build different DHLS compilers".

We could expose each compilation step as a standalone script (shell or Python), allowing both users and test cases to call them directly:

```
$ ./tools/dynamatic/scripts/compile (test folder)
$ ./tools/dynamatic/scripts/write-hdl (test folder)
$ ./tools/dynamatic/scripts/simulate (test folder)
```

For speculation, we'd provide a dedicated script: `$ ./tools/dynamatic/scripts/compile-spec (test folder)`

And for convenience, we can also provide a “run-all” wrapper:

```
$ ./tools/dynamatic/scripts/run-all (folder)
$ ./tools/dynamatic/scripts/run-all-spec (folder)
```

---

However, I'll emphasize the main point of this issue is **not** to decide whether `./bin/dynamatic` should be kept or removed.
The core concern is **how** to integrate custom compilation flows like speculation's one into CI and test suites while preserving usability.
More specifically, it is just about how to implement `TEST_P` for speculation integration tests.
The point around `./bin/dynamatic` is raised here only to highlight a related concern.

What do you think? @ebosnjak @schilkp (CC: @murphe67)

ebosnjak, 2025-04-07:
Although the current basic implementation of the integration tests directly calls `bin/dynamatic` (like the python script did before), you should be able to implement custom flows without any issues. For example, currently the tests contain:
```c++
TEST_P(BasicIntegrationFixture, basicNoFlags) {
  fs::path testPath = GetParam();
  EXPECT_EQ(runIntegrationTest(testPath), 0);
}
```

For a custom compilation flow, you would just add another parameterized test like this one and replace `runIntegrationTest` (which just calls `bin/dynamatic`) with your magic function, which can be either calling your `run_spec_integration.py` (option 2 in your post) or doing the same directly from C++. 

The only thing I would be careful about here is that by calling `run_spec_integration.py` results in more ~spaghetti~ lasagna (?), in the sense that we have a C++ test which calls a Python script which calls the dynamatic_opt binary and other tools. I do not think that this would be too bad for now (since rewriting the Python script is some work), but just something to keep in mind to prevent this scripts on scripts on scripts on executables architecture from escalating too much.

But the bottom line is that, as far as I can tell, what you described does not present any issue for the testing/CI flow and for me, since whatever crazy magic custom flow is needed it can just be added as a separate test. 

On the other hand, the question of the actual user experience with Dynamatic seems much more troublesome. I do not really have any good ideas regarding this matter, in fact, it seems to me that Dynamatic is one of these tools that will by design be plagued by having 10000 command line flags, options etc. regardless of how we try to avoid it. I mean, as a user, `bin/dynamatic` seems like a decent tool which allows basic usage without too much fuss, so I think that even if it is decided to be thrown out, there should still be some way to go from C to VHDL in such a simple way. Because, just like when compiling C in our everyday lives, we do not have to call the preprocessor then compile to assembly then assemble then call the linker, but we can do it all in one shot by just invoking gcc, I believe Dynamatic should be able to do the same regardless of all advanced features that are being added to it.

shundroid, 2025-04-07:
Thank you for your comment! Then I think I'll go with the option 1 (maybe I'll rename `run_spec_integration.py` with `run_spec_flow.py`).

> The only thing I would be careful about here is that by calling run_spec_integration.py results in more ~spaghetti~ lasagna (?), in the sense that we have a C++ test which calls a Python script which calls the dynamatic_opt binary and other tools.

That's a fair point, and I agree it could be a bit messy. But I don't think the number of lasagna layers is any worse than that of the normal test cases, considering what's inside bin/dynamatic:

- Normal test cases: C++ test -> bin/dynamatic -> shell scripts -> bin/dynamatic-opt
- Speculation test cases: C++ test -> run_spec_integration.py -> bin/dynamatic-opt

Also, while I agree that providing a one-shot command is important, I'm not sure bin/dynamatic fully qualifies. Getting RTL still requires three separate steps: (1) set-src, (2) compile, and (3) write-hdl, which can be combined in most cases.

schilkp, 2025-04-08:
Hi All!

Thank you for taking the time to write this out and your thoughts, @shundroid @ebosnjak.

> In my view, this issue is connected to a broader design concern with ./bin/dynamatic

@paolo-ienne , in the integration test system discussion, mentioned that the `/bin/dynamatic` shell was modeled after shell typical for EDA tools:

>  Years ago we took inspiration from the shells of classic EDA tools (I think we looked at Synopsys's Design Compiler in particular)

Those tools are typically based on some form of scripting language (usually tcl).

Much like my view of Dynamatic, these EDA tools are a "bag full of parts that I can use to build a synthesis/PnR/LVS flow".
The different parts are exposed as TCL functions, and the user can define a "flow" by either manually calling these functions in the CLI or writing a TCL script that performs the full end-to-end flow.
If I want to share a "preset" flow for a specific use case, I can just share the TCL script that performs that task.
Because TCL is a full scripting language, I can implement basic logic, error handling, functions, etc.

In my eyes, the issue is that `/bin/dynamatic` is only a first-order approximation of these tools, because it is not a full scripting engine. Instead, we rely on bash as the scripting language but this is not really exposed to the user.

Dynamatic sits somewhere in between an EDA tool and a compiler/software tool. At least for the project user interface, it would seem sensible to pick one of those two "styles" and implement it completely instead of the current "half EDA tool and half MLIR compiler" way. That gives us two options:

# Option 1: The EDA way

The first option would be to actually "upgrade" `/bin/dynamatic` to a full TCL interpreter, just like the EDA tools that inspired it.

Roughly speaking, this would require:
- Adding a TCL interpreter to `/bin/dynamatic`
- Exposing the different dynamatic "pieces"/"tools"/clis we have as built-in dynamatic functions
- Provide example "higher level" tcl functions that combine some of these low-level pieces into logical blocks.
- For each of our supported "flows" (vanilla, spec, ..) provide a TCL script that puts together the right combination of TCL functions/built-ins to implement that flow.

This would be a bit of work but, in my eyes, is not a horrible solution. I did not mind the TCL workflow when I was doing VLSI stuff.

# Option 2: The MLIR/XLS way

The second option would be to develop an interface more similar to what MLIR projects (and other software-style projects like XLS) expose, which is simply a whole set of CLI tools that the user can combine.

In general my opinion is that if `./bin/dynamatic` is just a tool that allows me to call shell scripts, it does not provide much value. I guess this aligns pretty much with what @shundroid wrote above.

I personally would:
- Remove `/bin/dynamatic`, and instead think of dynamatic as family of different CLI tools that each do one thing.
- Provide "example flows" that show how to put the different pieces together to achieve something specific.

These example flows would then, likely, be bash scripts.

For me, the big question is how to organize these scripts.

If at all possible, I would like the scripts to be as simple as possible (not a million options), have similar interfaces, and for compatibility of features to be immediately obvious.

> For speculation, we'd provide a dedicated script: $ ./tools/dynamatic/scripts/compile-spec (test folder)

I totally agree with the notion that speculation should be it's own script (or "flow", as I have been calling them), I just don't love the idea of having multiple `compile` scripts.

As far as I understand, your spec work has some very specific requirements for the backend/HDL generator and is not compatible with the verilog backend (correct me if I am wrong - but I will assume this and use it as an example from this point on). 

This means that if I use `compile-spec`, I have to use `write-hdl` with very specific flags otherwise it breaks. 
This is annoying, stuff we would have to document, and gets confusing for testing because it is no longer clear which possible combinations of parameters are valid.

We currently have two types of scripts:
- "Compilation Scripts" that actually form part of the dynamatic flow (`compile` and `write-hdl`)
- "Testing Scripts" that take the dynamatic flow and do something with it (`synthesize`, `simulate`, `visualize`)

Looking at `write-hdl`, it is super simple.

How about, we instead do the following:
- "Flows", which are end-to-end "recipes" for running dynamatic (`dynamatic`,  `dynamatic-spec`,  `dynamatic-foobar`, ...).
- "Testing scripts" as above.

In short, this means pulling `write-hdl` into the `compile` script and making a new compile script for every flow.

These flow scripts then take, as arguments, exactly the set of parameters that can be changed, and hard-code everything else.

For example, the `dynamatic` script would take the following options (next to input/outputs/names):
```
--use_sharing=true|false
--buffer_algo=on-merges|...
--rtl=verilog|vhdl
```

In turn, the `dynamatic-spec` script would take a much smaller set of arguments, and always emit vhdl:
```
--use_sharing=true|false
--buffer_algo=on-merges|...
```

I like this because it is much easier for the user to understand, and also makes it clear what permutations of parameters we actually support. The later is super critical for us to define, because it directly tells us what we should be testing:
- for `dynamatic` we have every possible (`use_sharing`, `buffer_algo`, `rtl`) triple that should be tested
- for `dynamatic-spec` we should test only every possible (`use_sharing`, `buffer_algo`)

What do you think?

schilkp, 2025-04-08:
Some more thoughts about how we write these "flows".

The obvious choice is a shell script, but I don't love that idea.

Currently, `compile.sh` and `write-hdl` combined require the following args:

```
DYNAMATIC_DIR
SRC_DIR
OUTPUT_DIR
KERNEL_NAME
BUFFER_ALGORITHM
TARGET_CP
POLYGEIST_PATH
USE_SHARING
HDL
```

This makes it very annoying to use, since all of these have to be specified.

There are some "tricks" we can play to get rid of some of these:
- We could require `DYNAMATIC_DIR` to be an env variable, or the user to add the folder to their `$PATH`, or a combination of "either have the `DYNAMATIC_DIR` env var defined, or give the --dynamatic_dir=... flag"
- Same for `POLYGEIST_PATH`
- We could have `SRC_DIR` default to the current directory unless a `--src_dir` flag is given.
- We could have `OUTPUT_DIR` default to `SRC_DIR/out`
- ...

All this adds up to a whole bunch of annoying argument and path parsing inside these scripts, which is error prone and confusing to read - especially in bash. Just have a look at `build.sh` :)

Some possible solutions:

### Do like gcc/clang - Have `dynamatic`, `dynamatic-spec` etc be driver programs

The `gcc` executable is not actually the compiler, but just a driver that calls all the correct sub-tools with the correct flags for the given options - the preprocessor, `cc1` (the actual compiler), the assembler, the linker.

We could adopt the same structure for dynamatic. What if `dynamatic` and `dynamatic-spec` are just small C++ programs with proper arg parsing that call the relevant tools?

### Have `dynamatic`, `dynamatic-spec` be python scripts

I guess they could also be python scripts, but I would try to stick to one language as much as possible.

### Keep `./bin/dynamatic`

I guess we address most of our problems with `./bin/dynamatic` by changing `compile` and `write-hdl`  into "end-to-end flows", is it worth just keeping it around for the path management stuff?

---

Think aloud - please chime in :)

shundroid, 2025-04-10:
Thanks for sharing your idea, and sorry for my late reply!

I agree with your points on:

- Separating scripts into flow and testing categories
- Treating flows as end-to-end, including HDL export

Regarding the Verilog backend not supporting speculation, I think that's rather a temporary limitation. More precisely, I would say it doesn't support the extra signal feature in the IR (which is a more general mechanism also used by out-of-order execution). So to me, the compatibility issue is more about the relationship between an IR feature and individual pass, rather than between passes (which I guess is actually a common issue that exists in many passes).

That said, I still think it's user-friendly to provide flows all the way to HDL export. Also exposing the weird `write-hdl` customization to end-users doesn't feel like a great idea.

As for managing flows, I personally think something gcc/clang-like would suit us the most. Maybe we could provide something like a `dcc` or `dcc-spec` tool, which takes a C file (or project directory) as input and generates an out directory with `comp` and `hdl` folder.

This is a bit of a technical detail, but if we implement these in C++ instead of Python, maybe we can reuse the pass manager feature from MLIR? Not sure how helpful it'd actually be though.

schilkp, 2025-04-11:
> Regarding the Verilog backend not supporting speculation, I think that's rather a temporary limitation. More precisely, I would say it doesn't support the extra signal feature in the IR (which is a [...]

Got it.

> As for managing flows, I personally think something gcc/clang-like would suit us the most. Maybe we could provide something like a dcc or dcc-spec tool, which takes a C file (or project directory) as input and generates an out directory with comp and hdl folder.

Agreed. Thinking about it more, I think that I also like this option the best.

> This is a bit of a technical detail, but if we implement these in C++ instead of Python, maybe we can reuse the pass manager feature from MLIR? Not sure how helpful it'd actually be though.

Absolutely. We currently do "pass management" in the shell script by running one pass at a time, but it totally makes sense to just keep everything in-memory and use the MLIR pass infrastructure. The current approach that writes every step to a file is nice for debugging, but not really super efficient. But as long as the "driver programs" don't do any actual business logic but just chain MLIR passes and external tools together, it is still possible to do everything by hand manually during development. Also, nothing says that we can't dump every stage to a file in our driver program (possibly gated behind a flag).

schilkp, 2025-04-11:
Another benefit of going C++:

We can link against the drive program directly from our gtest suite, instead of calling the executable. 

(This is also how google does it for XLS, I might add in character).

The way that XLS structures it, is that the actual "driver code" that sequences the compilation stuff is in a lib, and there is a small C++ executable that just does CLI arg parsing and calls this lib.

The tests can directly call the lib.

schilkp, 2025-04-11:
In general, I think that this is shaping up to be a sufficiently large refactor, that it would be good to have a discussion about it in a Dynamatic meeting :)

Happy to help out putting a short presentation together for that, if there is interest and OK for everyone.