EPFL-LAP/dynamatic #382: [Compile Script] Export .png for CFGs

AyaElAkhras, 2025-04-11:
It would be useful to extend the compilation script to produce a `.png` file of the control flow graphs for the benchmarks being compiled.

This would save time for anyone interested in understanding the control flow structure of the kernel under study.

I believe it can be implemented similarly to how the Handshake IR is currently exported to DOT format and then converted to a .png.

Below is an example of what I have in mind, taken from the legacy LLVM-based Dynamatic using the kernel_2mm benchmark.

[image]

Jiahui17, 2025-04-19:
Should this be generated from the CF dialect or handshake?

AyaElAkhras, 2025-04-22:
> Should this be generated from the CF dialect or handshake?

From the CF dialect because currently we flatten the CFG during the CF to handshake conversion so it will be easier to generate it while still having the structure at the CF level.

PR #404 (2025-04-30): [tools] export-cfg: Export *_CFG.png
Example (polyn_mult):

[image]

fixes #382
commit da016a1a (2025-04-30): [tools] export-cfg: Export *_CFG.png (#404)

Example (polyn_mult):

[image]

fixes #382
closed 2025-04-30 by AyaElAkhras