EPFL-LAP/dynamatic #383: [transformations to handshake] compile  error in my backpropagation project

lyl-seu, 2025-04-12:
Benchmark:(bp.c is successful, but bp_2.c failed)
[bp.zip](https://github.com/user-attachments/files/19718449/bp.zip)
### Description
When I add the following code in `bp.c`.
`for (int j = 0; j < HIDDEN_SIZE; j++) { `
                `for(int k = 0; k < OUTPUT_SIZE; k++) {`
                   ` W2[j][k] += LEARNING_RATE * hidden[j] * output_delta[k];`
               ` }`
          `  }`

It encountered the same issue as #133.So ,i pass the local array as a function argument instead ,as the #133 says.And the new problem occurs in my `bp_2.c`.

`dynamatic/scripts/compile.sh: line 125:  4109 Aborted                 (core dumped) "$DYNAMATIC_OPT_BIN" "$F_HANDSHAKE" --handshake-analyze-lsq-usage --handshake-replace-memory-interfaces --handshake-minimize-cst-width --handshake-optimize-bitwidths --handshake-materialize --handshake-infer-basic-blocks > "$F_HANDSHAKE_TRANSFORMED"
[FATAL] Failed to apply transformations to handshake`

I'm a newbie in this field and I really hope to get your help. Thank you so much!

Jiahui17, 2025-04-14:
I completely forgot #133 ... Have you pulled all the local array into the argument? It looks like you still have 3 local arrays in bp_2.c:

[image]

lyl-seu, 2025-04-17:
Thank you for your reply @Jiahui17 . In the `bp_2.c` that I provided, I did not move all the local arrays into function variables because I wanted to keep the differences between `bp_2` and `bp_1` as minimal as possible to help you identify the issue more easily. In fact, I also tried passing all the local arrays as function arguments, but this resulted in another error:

```
<unknown>:0: error: expected data type to have strictly positive bitwidth, but got 'i0'
static ConcreteT mlir::detail::StorageUserBase<dynamatic::handshake::ChannelType, mlir::Type, dynamatic::handshake::detail::ChannelTypeStorage, 
mlir::detail::TypeUniquer>::get(MLIRContext *, Args...) [ConcreteT = dynamatic::handshake::ChannelType, BaseT = mlir::Type, StorageT = dynamatic::handshake::detail::ChannelTypeStorage, UniquerT = mlir::detail::TypeUniquer, Traits = <>, Args = <mlir::Type, llvm::ArrayRef<dynamatic::handshake::ExtraSignal>>]: Assertion `succeeded(ConcreteT::verify(getDefaultDiagnosticEmitFn(ctx), args...))' failed.
```
maybe Issue like  #336,i don't konw how to do with it.QvQ

lyl-seu, 2025-05-07:
@Jiahui17 thanks Jiahui,You truly solved my problem and my Compilation succeeded!
while in next`write-hdl`step,information are below:
`Uncaught module: handshake.lsq`
`Uncaught module: handshake.lazy_fork`
`[INFO] Exported RTL (vhdl)`
`[INFO] HDL generation succeeded`

PR #415 (2025-05-08): [HDL][Bitwidth] Workarounds for 1-bit array
- Workarounds for 1-bit or 1-element arrays in RTL gen.
- Disable bitwidth optimization for left shift (if left shift is used as a means for reducing the arithmetic strength of a multiplication, we shouldn't shrink the bitwidth).

fixes #383
commit 80566d6c (2025-05-08): [HDL][Bitwidth] Workarounds for 1-bit array (#415)

- Workarounds for 1-bit or 1-element arrays in RTL gen.
- Disable bitwidth optimization for left shift (if left shift is used as
a means for reducing the arithmetic strength of a multiplication, we
shouldn't shrink the bitwidth).

fixes #383