EPFL-LAP/dynamatic #442: [Build] Build fails on Arch Linux due to polygeist/llvm-project

ebosnjak, 2025-05-18:
I have been unable to build Dynamatic at all in the last two weeks due to a bunch of errors in polygeist/llvm-project.

It is not a problem with my machine, I also tried building Dynamatic in a Docker container (using image `archlinux:base-devel-20250511.0.348143` and following the build instructions in the readme) and the result was the same: a bunch of errors related to types such as `uint32_t` being undefined
```
/home/eugenb/dynamatic/polygeist/llvm-project/llvm/include/llvm/ADT/SmallVector.h:109:62: error: use of undeclared identifier 'uint64_t'
  109 |     std::conditional_t<sizeof(T) < 4 && sizeof(void *) >= 8, uint64_t,
```

Even if I manually added `#include <cstdint>` to all of the problematic files, I would then get other unrelated errors. I tried changing C++ version from 17 to 14, but that resulted in the build failing because C++17 features are required.

I know that this is not a Dynamatic issue, but I think that we should update the project readme which currently states that it works "_out-of-the-box for any apt-based or pacman-based Linux distribution_"; the latter is not true.

ebosnjak, 2025-06-09:
@pcineverdies Unfortunately not; I resorted to building and running Dynamatic in an Ubuntu Docker container with the help of VS Code’s Dev Containers extension.

But if it doesn’t work on Fedora either, that’s pretty sus now (I would otherwise just write it off as “arch is rolling release so problems happen”). Can you send me the list of versions of all (possibly) relevant stuff that you are using? Like clang, libc++ etc. so maybe I can compare it and try to figure out who is the culprit.

pcineverdies, 2025-06-10:
Asap I'll send you everything. 
Btw, @schilkp might solve a doubt of mine here: can this be related to the old version of LLVM we are relying on?

pcineverdies, 2025-06-10:
Thanks Philip! Very informative answer!!
I'll downgrade then...

ebosnjak, 2025-06-10:
@schilkp Thanks for the update! I patched the project so that llvm project is replaced with your version and the build now succeeds natively on my machine. However, the problem is that the integration test `kmp` fails with the following error output:

```
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.	Program arguments: /home/eugenb/tmp/dynamatic/bin/dynamatic-opt /home/eugenb/tmp/dynamatic/integration-test/kmp/out/comp/handshake.mlir --handshake-analyze-lsq-usage --handshake-replace-memory-interfaces --handshake-minimize-cst-width --handshake-optimize-bitwidths --handshake-materialize --handshake-infer-basic-blocks
 #0 0x000055746aff5546 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0x3ac546)
 #1 0x000055746aff30fe llvm::sys::RunSignalHandlers() (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0x3aa0fe)
 #2 0x000055746aff5ef4 SignalHandler(int) Signals.cpp:0:0
 #3 0x00007f154924def0 (/usr/lib/libc.so.6+0x3def0)
 #4 0x000055746b6ec537 std::_Function_handler<(anonymous namespace)::GIIDStatus (mlir::Value), isGIIDRec(mlir::Value, mlir::OpOperand&, llvm::ArrayRef<unsigned int>)::$_0>::_M_invoke(std::_Any_data const&, mlir::Value&&) CFG.cpp:0:0
 #5 0x000055746b6ead65 isGIIDRec(mlir::Value, mlir::OpOperand&, llvm::ArrayRef<unsigned int>) CFG.cpp:0:0
 #6 0x000055746b6ea23f dynamatic::isGIID(mlir::Value, mlir::OpOperand&, llvm::SmallVector<unsigned int, 12u>&) (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0xaa123f)
 #7 0x000055746b1aee3c (anonymous namespace)::HandshakeAnalyzeLSQUsagePass::analyzeFunction(dynamatic::handshake::FuncOp) HandshakeAnalyzeLSQUsage.cpp:0:0
 #8 0x000055746b1ad7b3 (anonymous namespace)::HandshakeAnalyzeLSQUsagePass::runDynamaticPass() HandshakeAnalyzeLSQUsage.cpp:0:0
 #9 0x000055746b015c78 dynamatic::DynamaticPass::runOnOperation() AffineToScf.cpp:0:0
#10 0x000055746b94183f mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0xcf883f)
#11 0x000055746b945b19 mlir::PassManager::run(mlir::Operation*) (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0xcfcb19)
#12 0x000055746b5aa0ba performActions(llvm::raw_ostream&, std::shared_ptr<llvm::SourceMgr> const&, mlir::MLIRContext*, mlir::MlirOptMainConfig const&) MlirOptMain.cpp:0:0
#13 0x000055746b5a9190 mlir::LogicalResult llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>::callback_fn<mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::DialectRegistry&, mlir::MlirOptMainConfig const&)::$_0>(long, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) MlirOptMain.cpp:0:0
#14 0x000055746b5c5d90 mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool) (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0x97cd90)
#15 0x000055746b5a5487 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::DialectRegistry&, mlir::MlirOptMainConfig const&) (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0x95c487)
#16 0x000055746b5a591c mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&) (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0x95c91c)
#17 0x000055746afc2f55 main (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0x379f55)
#18 0x00007f15492376b5 __libc_start_call_main /usr/src/debug/glibc/glibc/csu/../sysdeps/nptl/libc_start_call_main.h:74:3
#19 0x00007f1549237769 call_init /usr/src/debug/glibc/glibc/csu/../csu/libc-start.c:128:20
#20 0x00007f1549237769 __libc_start_main /usr/src/debug/glibc/glibc/csu/../csu/libc-start.c:347:5
#21 0x000055746afc2b35 _start (/home/eugenb/tmp/dynamatic/bin/dynamatic-opt+0x379b35)
/home/eugenb/tmp/dynamatic/tools/dynamatic/scripts/compile.sh: line 143: 90363 Segmentation fault      (core dumped) "$DYNAMATIC_OPT_BIN" "$F_HANDSHAKE" --handshake-analyze-lsq-usage --handshake-replace-memory-interfaces --handshake-minimize-cst-width --handshake-optimize-bitwidths --handshake-materialize --handshake-infer-basic-blocks > "$F_HANDSHAKE_TRANSFORMED"
```

AyaElAkhras, 2025-06-10:
> If this does fix it, I am happy to get this setup in dynamatic. I just would need someone to fork both LLVM and Polygeist into the EPFL-LAP github group and give me access - I don't think I have permission.
> 
> We can also discuss on thursday :)

I'm about to send an email canceling this Thursday's meeting because of IWLS :/ Maybe the best is to email Paolo and Lana to explain the situation and request access?

schilkp, 2025-06-10:
> [@schilkp](https://github.com/schilkp) Thanks for the update! I patched the project so that llvm project is replaced with your version and the build now succeeds natively on my machine. However, the [...]

Hi!

I added an extra upstream patch that fixed some undefined behavior related to SmallVector to my branch.

However, both with and without that extra patch, I cannot reproduce your crash :/

On my local machine (GCC v15, gurobi, but no modelsim), I can build everything just fine and `kmp` integration test compiles - but fails because there is no vsim ofc:

```
[INFO] Compiled source to affine
[INFO] Ran memory analysis
[INFO] Compiled affine to scf
[INFO] Compiled scf to cf
[INFO] Applied standard transformations to cf
[INFO] Applied Dynamatic transformations to cf
[INFO] Compiled cf to handshake
[INFO] Applied transformations to handshake
[INFO] Running simple buffer placement (on-merges).
[INFO] Placed simple buffers
[INFO] Canonicalized handshake
[INFO] Created kmp DOT
[INFO] Converted kmp DOT to PNG
[INFO] Created kmp_CFG DOT
[INFO] Converted kmp_CFG DOT to PNG
[INFO] Lowered to HW
[INFO] Compilation succeeded
[INFO] Exported RTL (vhdl)
[INFO] HDL generation succeeded
[INFO] Built kernel for IO gen.
[INFO] Ran kernel for IO gen.
[INFO] Launching Modelsim simulation
[FATAL] Simulation failed

[INFO  HLS_VERIFIER] Executing modelsim: [vsim -c -do simulation.do]
sh: line 1: vsim: command not found
[ERROR UTIL] Execution failed for command [vsim -c -do simulation.do]
hls-verifier: /home/schilkp/st_hls/dynamatic/tools/hls-verifier/lib/Utilities.cpp:198: bool executeCommand(const string &): Assertion `false && "Command execution failed!"' failed.
```

On the ETH server (gcc v8.5, modelsim + gurobi), everything works both with and without my patches, including all integration tests.

Unfortunately I don't easily have access to an environment with gcc v15, modelsim, and gurobi at the moment.

When you have a minute, could you do a *clean* build against my llvm-project branch? 

> [@schilkp](https://github.com/schilkp) You are listed as a team member [here](https://github.com/orgs/EPFL-LAP/people), are you sure you don't have access? I mean, I might be missing something but seems that you should be able to add stuff.

Ah I had missed the invite - Thanks.

Is it OK for you all if I fork LLVM + polygeist into your org?

schilkp, 2025-06-11:
> @schilkp I tried it in a Docker with image archlinux:base-20250608.0.361578⁠ and I get the same error, so I guess that's one way to reproduce it.

I have done the same and cannot repro the crash. For the record, here is what I am doing:

```dockerfile
FROM archlinux:base-20250608.0.361578

# Install deps
RUN pacman -Syu --noconfirm && \
    pacman -S --noconfirm \
    clang \
    lld \
    ccache \
    cmake \
    ninja \
    python \
    jdk21-openjdk \
    graphviz \
    git \
    curl \
    gzip \
    readline \
    base-devel \
    which \
    unzip \
    bash \
    openssh \
    sbt

# Install scala
RUN curl -fL https://github.com/coursier/coursier/releases/latest/download/cs-x86_64-pc-linux.gz \
    | gzip -d > cs && \
    chmod +x cs && \
    ./cs setup --env > ./add_to_path && \
    cat ./add_to_path && \
    cat ./add_to_path >> /etc/profile && \
    cat ./add_to_path >> /etc/bash.bashrc && \
    source ./add_to_path && \
    rm cs

# Clone dynamatic fork with polygeist fork with patched version of LLVM:
WORKDIR /opt
RUN git clone --recurse-submodules https://github.com/schilkp/dynamatic.git

# Log GCC version:
WORKDIR /opt/dynamatic
RUN gcc --version > ./gcc_version && \
    g++ --version > ./g++_version && \
    cc --version  > ./cc_version && \
    c++ --version  > ./c++_version
    
# Build
WORKDIR /opt/dynamatic
RUN chmod +x ./build.sh && \
    ./build.sh 

# Compile kmp int. test:
WORKDIR /opt/dynamatic
ARG CACHEBUST=1
RUN echo -e "set-src ./integration-test/kmp/kmp.c\ncompile" > ./run.dyn && \
    ./bin/dynamatic --run ./run.dyn

# Set default working directory
WORKDIR /opt/dynamatic
```

Inside empty directory, with just this Dockerfile, run:
```bash
docker build -t dynamatic-debug .
```

Then, walk around the block a few times, and come back to this:

```text
....
Step 12/13 : RUN echo -e "set-src ./integration-test/kmp/kmp.c\ncompile" > ./run.dyn &&     ./bin/dynamatic --run ./run.dyn
 ---> Running in 7954eea9b9d2
[INFO] Compiled source to affine
[INFO] Ran memory analysis
[INFO] Compiled affine to scf
[INFO] Compiled scf to cf
[INFO] Applied standard transformations to cf
[INFO] Applied Dynamatic transformations to cf
[INFO] Compiled cf to handshake
[INFO] Applied transformations to handshake
[INFO] Running simple buffer placement (on-merges).
[INFO] Placed simple buffers
[INFO] Canonicalized handshake
[INFO] Created kmp DOT
[INFO] Converted kmp DOT to PNG
[INFO] Created kmp_CFG DOT
[INFO] Converted kmp_CFG DOT to PNG
[INFO] Lowered to HW
[INFO] Compilation succeeded
dynamatic> ================================================================================
============== Dynamatic | Dynamic High-Level Synthesis Compiler ===============
======================== EPFL-LAP - v2.0.0 | March 2024 ========================
================================================================================
```

Note that this clones from a fork of dynamatic, which has a fork of polygeist as a submodule, which in turn has my fork + branch of LLVM as a submodule.

Same behavior on my local machine, ETH server, ...

Could you please share the exact steps you are doing to create this crash? Environment? Commands? Dockerfiles? Pre-existing build state? What are you running to trigger the crash?

schilkp, 2025-06-12:
@ebosnjak @pcineverdies 

Could you please try and build/test https://github.com/schilkp/dynamatic (main), especially in the environment/with the GCC version that previously failed? 

On my machine + arch docker, this fixes it - both in release and debug :)

Note that this has different sub modules, so just pulling from there won't directly work :)

commit 0bcbe640 (2025-06-25): Update polygeist/LLVM to pick-up backported GCCv15 compat. fixes. (#500)

Fixes #442.

The version of LLVM dynamatic uses is the same version used by
Polygeist, which is a few years old.

As different linux distros have started shipping the new GCC v15, the
project no longer compiled correctly. This is because of some changes in
GCC's standard library headers, which used to include stdint.h (or
cstdint), but no longer do. Any files that relied on fixed-width
integers without themselves including the stdint may have compiled fine
with previous versions but fail with GCC v15.

https://gcc.gnu.or [... cut]