EPFL-LAP/dynamatic #507: Simulate command on Frontend Dynamatic not working.

hoodude717, 2025-06-27:
I am working through the Introductory tutorials for Dynamatic and in the first example I follow everything exactly and this issue keeps coming up with the simulate. 

The following is the terminal output from the tutorial

```dynamatic> set-src tutorials/Introduction/Ch1/loop_multiply.c
dynamatic> compile
[INFO] Compiled source to affine
[INFO] Ran memory analysis
[INFO] Compiled affine to scf
[INFO] Compiled scf to cf
[INFO] Applied standard transformations to cf
[INFO] Applied Dynamatic transformations to cf
[INFO] Compiled cf to handshake
[INFO] Applied transformations to handshake
[INFO] Running simple buffer placement (on-merges).
[INFO] Placed simple buffers
[INFO] Canonicalized handshake
[INFO] Created loop_multiply DOT
[INFO] Converted loop_multiply DOT to PNG
[INFO] Created loop_multiply_CFG DOT
[INFO] Converted loop_multiply_CFG DOT to PNG
[INFO] Lowered to HW
[INFO] Compilation succeeded
dynamatic> write-hdl
[INFO] Exported RTL (vhdl)
[INFO] HDL generation succeeded
dynamatic> simulate
[INFO] Built kernel for IO gen.
[INFO] Ran kernel for IO gen.
[INFO] Launching Modelsim simulation
[FATAL] Simulation failed

After looking through the reports I find this in the report.txt in the out/sim directory

[INFO  HLS_VERIFIER] Executing modelsim: [vsim -c -do simulation.do]
/opt/modelsim_ase/bin/../linux/vish: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
[ERROR UTIL] Execution failed for command [vsim -c -do simulation.do]
[ERROR UTIL] Output file does not exist: /home/hoodoo17/Desktop/Dynarapid/dynamatic/tutorials/Introduction/Ch1/out/sim/HDL_OUT/output_a.dat
Comparison of [a] : Fail
[ERROR HLS_VERIFIER] C and VHDL outputs do not match ```

hoodude717, 2025-06-30:
@shundroid  and @A-Alosius Thank you for responding. When I try and download the libncurses.so.5 library it says there is no libncurses.so.5 package I can install. I have been trying to work around it but it seems necessary for the modelsim. vsim just says the following.

`/opt/modelsim_ase/bin/../linux/vish: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory`

A-Alosius, 2025-06-30:
did you enable the architecture i386 as in my sample code line 1?

A-Alosius, 2025-06-30:
Check this link out.
It highlights a number of dependencies that Modelsim usually requires beyond the ones we suggested

https://pcotret.github.io/modelsim-ubuntu/

Let me know if it helps

hoodude717, 2025-07-01:
@A-Alosius I followed all the instructions in the website you sent. Still not working.
New Error: /opt/modelsim_ase/bin/../linux/vish: symbol lookup error: /lib/i386-linux-gnu/libfontconfig.so.1: undefined symbol: FT_Done_MM_Var

I do have a question, is there no other newer architecture for simulation that Dynamatic could use? If the modelsim is 32bit it requires a lot of work around to get it to work and even then doesnt always work.

A-Alosius, 2025-07-04:
Thank you for the feedback. I just created a dual boot with ubuntu and built Dynamatic with the most recent version of Questa sim (Modelsim's successor). I haven't installed any i386 package and it works. Try removing Modelsim and using Questa and let me know if it works for you too

A-Alosius, 2025-07-22:
> Thank you for the feedback. I just created a dual boot with ubuntu and built Dynamatic with the most recent version of Questa sim (Modelsim's successor). I haven't installed any i386 package and it works. Try removing Modelsim and using Questa and let me know if it works for you too

@hoodude717 did you get to try Questa?