EPFL-LAP/dynamatic #529: [Buffering] Suboptimal Throughput with Memory Store (FPGA20, FPL22)

shundroid, 2025-07-13:
The buffering algorithm reports the program below can achieve a throughput of 1.0:

```
# ================= #
# CFDFC Throughputs #
# ================= #

Throughput of CFDFC #0: 1.000000e+00
```

However, the actual observed throughput is only 1/6, significantly lower than the reported value.
This discrepancy occurs with both the FPGA20 and FPL22 algorithms.

When I manually insert buffers, the design achieves the expected throughput of 1.0.

It seems that buffers before memory stores are being skipped. However, the MLIP results still suggest not placing buffers in these locations.

I’m currently investigating the issue.

Program:

```c
#include "test_loop.h"
#include "dynamatic/Integration.h"
#include <stdlib.h>

#define N 100
void test_loop(int a[N], int b[N]) {
  for (int i = 0; i < N; i++) {
    b[i] = a[i] * i;
  }
}

int main(void) {
  int a[N];
  int b[N];

  srand(13);
  for (int j = 0; j < N; j++) {
    a[j] = rand() % 100;
  }

  CALL_KERNEL(test_loop, a, b);
  return 0;
}
```

QinYuan2000, 2025-07-13:
The throughput result in MILP does not necessarily reflect the simulation result. Due to undetected and unsolved problems, some benchmarks could be a bit degraded in the actual circuit compared to the optimization model. Still, a difference of 1/6 is too big to be reasonable.

QinYuan2000, 2025-07-13:
> It seems that buffers before memory stores are being skipped.

I only know we used these lines to place these buffers. They should have worked previously.

https://github.com/EPFL-LAP/dynamatic/blob/d5549d0ddee62418ad590b8876f71b08aec9eda1/lib/Transforms/BufferPlacement/HandshakeSetBufferingProperties.cpp#L127-L145

Jiahui17, 2025-07-13:
@QinYuan2000 Aren't we always placing transparent buffers before the stores?

Jiahui17, 2025-07-14:
@shundroid could you annotate the reconvergent path that leads to this throughput drop?

closed 2025-11-05 by Jiahui17