EPFL-LAP/dynamatic #547: [Frontend] Condition of BB3 defined in BB2

AyaElAkhras, 2025-07-23:
The `iterative_sqrt` integration test https://github.com/EPFL-LAP/dynamatic/tree/main/integration-test/iterative_sqrt crashes with the fast token delivery conversion pass. 

After some debugging, I came to conclude that the problem arises from the frontend. Precisely, the CF MLIR has the value representing the condition of BB3 calculated inside BB2, as indicated by the value `%19` in the following `cf_dyn_transformed.mlir`, which is the input to the CfToHandshake conversion pass: 

```
module {
  func.func @iterative_sqrt(%arg0: i32 {handshake.arg_name = "n"}) -> i32 {
    %c0_i32 = arith.constant {handshake.name = "constant0"} 0 : i32
    %true = arith.constant {handshake.name = "constant3"} true
    %0 = llvm.mlir.undef {handshake.name = "mlir.undef0"} : i32
    cf.br ^bb1(%arg0, %c0_i32, %true, %0, %true : i32, i32, i1, i32, i1) {handshake.name = "br0"}
  ^bb1(%1: i32, %2: i32, %3: i1, %4: i32, %5: i1):  // 4 preds: ^bb0, ^bb2, ^bb4, ^bb5
    %6 = arith.cmpi sle, %2, %1 {handshake.name = "cmpi0"} : i32
    %7 = arith.andi %6, %5 {handshake.name = "andi0"} : i1
    cf.cond_br %7, ^bb2(%1, %3, %4, %2 : i32, i1, i32, i32), ^bb6 {handshake.name = "cond_br0"}
  ^bb2(%8: i32, %9: i1, %10: i32, %11: i32):  // pred: ^bb1
    %c1_i32 = arith.constant {handshake.name = "constant4"} 1 : i32
    %12 = arith.addi %11, %8 {handshake.name = "addi0"} : i32
    %13 = arith.shrsi %12, %c1_i32 {handshake.name = "shrsi0"} : i32
    %14 = arith.muli %13, %13 {handshake.name = "muli0"} : i32
    %15 = arith.cmpi eq, %14, %arg0 {handshake.name = "cmpi1"} : i32
    %16 = arith.cmpi ne, %14, %arg0 {handshake.name = "cmpi2"} : i32
    %17 = arith.andi %16, %9 {handshake.name = "andi1"} : i1
    %18 = arith.select %15, %13, %10 {handshake.name = "select0"} : i32
    %19 = arith.cmpi slt, %14, %arg0 {handshake.name = "cmpi3"} : i32
    cf.cond_br %16, ^bb3, ^bb1(%8, %11, %17, %18, %16 : i32, i32, i1, i32, i1) {handshake.name = "cond_br1"}
  ^bb3:  // pred: ^bb2
    cf.cond_br %19, ^bb4, ^bb5 {handshake.name = "cond_br2"}
  ^bb4:  // pred: ^bb3
    %c1_i32_0 = arith.constant {handshake.name = "constant5"} 1 : i32
    %20 = arith.addi %13, %c1_i32_0 {handshake.name = "addi1"} : i32
    cf.br ^bb1(%8, %20, %17, %18, %16 : i32, i32, i1, i32, i1) {handshake.name = "br1"}
  ^bb5:  // pred: ^bb3
    %c-1_i32 = arith.constant {handshake.name = "constant6"} -1 : i32
    %21 = arith.addi %13, %c-1_i32 {handshake.name = "addi2"} : i32
    cf.br ^bb1(%21, %11, %17, %18, %16 : i32, i32, i1, i32, i1) {handshake.name = "br7"}
  ^bb6:  // pred: ^bb1
    %22 = arith.select %3, %1, %4 {handshake.name = "select1"} : i32
    return {handshake.name = "return0"} %22 : i32
  }
}
```

This causes a major problem to fast token delivery because it is based on studying the conditions of production and consumption by looking at the BBs of the operations. Currently, it sees the condition of BB3 (`c3`) as no different from the condition of BB2 (`c2`), which affects the control dependence analysis. Happy to give more details about the concrete example of dependencies where the problem arises, if it helps.

Here is a visualization of the CFG for reference. `c2` and `c3` should be treated separately.

[image]

@Jiahui17, do you think the progress made on fixing the frontend issues by translating from affine to llvm can help overcome this problem? Please feel free to tag more people working on this.

Jiahui17, 2025-07-23:
So the analysis assumes that the condition must be computed in the same block as it is used?

AyaElAkhras, 2025-07-23:
> So the analysis assumes that the condition must be computed in the same block as it is used?

The condition of one BB that decides where the control flow will go next should be inside that BB, yes. The fact that the condition deciding for BB3 is inside BB2 is problematic for the analysis.

AyaElAkhras, 2025-08-11:
Solved in #564.

closed 2025-08-11 by AyaElAkhras